bipropellant/bipropellant-hoverboard-api #4: ABout arduino

chronicle972, 2019-07-26:
i can figure out how Im suppose t use the arduino with , There is some Howto somewhere ?

p-h-a-i-l, 2019-07-27:
Sorry, no howto yet. Maybe I can walk you through it by answering your questions, and you could document your experience by putting it into a howto for others?

First, you'll need to bipropellant-hoverboard-firmware running on your board in a UART config.

chronicle972, 2019-07-27:
OK , im not that good at all of this , so I hope my questions wont be too stupid. 
1- I'm guessinng I need to compile it in a hex file then  put it on the board with the st link ? 
2- what do you mean by UART config ?

p-h-a-i-l, 2019-07-28:
Yes, you need to compile and flash the board.
The firmware can work with many different input methods, you need to specify it through precompiler switches. Usually in config.h.

verial, 2019-07-28:
I also did not work, what parameters to choose the gpio ?

p-h-a-i-l, 2019-07-28:
Could you elaborate a little what you tried and how it failed? What gpio?

verial, 2019-07-29:
I compiled your program as is and flashed the gyro controller go data only on PA2.

chronicle972, 2019-07-30:
I have this error when i try to put the api on the arduino :
fatal error: hbprotocol/protocol.h: No such file or directory

p-h-a-i-l, 2019-07-31:
Sorry, I'm not able to follow what you are doing. Which gyro controller? What about PA2?

closed 2019-11-05 by p-h-a-i-l