bipropellant/bipropellant-hoverboard-api #5: GPIO

verial, 2019-07-29:
I use SOFTWARE_SERIAL_A2_A3 control type configuration for my control board: 

#define CONTROL_TYPE SOFTWARE_SERIAL_A2_A3

What values should I set for SENSOR_BAUD, SENSOR_WORDS and SOFTWARE_SERIAL_BAUD?

p-h-a-i-l, 2019-07-29:
Do you have both Sensor wires already in use? If not, I'd recommend to use USART2 or USART3. I 
 see Software serial as a last resort when no hardware UART is available.

SENSOR_BAUD and SENSOR_WORDS are only needed when communicating to the original hoverboards, I've never used them.

I think SOFTWARE_SERIAL_BAUD should be at 9600, everything else might be too fast for bitbanging. 
When in doubt, try to go with the default.

verial, 2019-07-29:
I tested about 20 different gyroscuter's control boards. The boards assembled on two types of processors - STM and GD (GigaDevice Semiconductor).

When I'm use  SOFTWARE_SERIAL_A2_A3 control style:

STM processor boards work OK.

GD processor board don't work properly. Wheel or don't rotate at all, or twitching, or trying to rotate backward.

Can different processor frequencies be the reason for this behavior of the controller in the second case? GD processor core operation at 108MHz frequency.

verial, 2019-08-08:
example, you can adjust the config file to work with usart 3 pins PA2 and PA3 sketch on Arduino. Thanks

closed 2019-11-05 by p-h-a-i-l