bipropellant/bipropellant-hoverboard-api #6: Arduino PWM example doesn't work for me

ahmed380, 2019-08-03:
Not sure if I am doing something wrong. The example here did not work for me starting the wheels via pwm  [https://github.com/bipropellant/bipropellant-hoverboard-api/blob/master/examples/PWMsimple/PWMsimple.ino](url)

using ST-Link debugger, and in protocol_byte function, I can see that the message coming from Arduino is aligned and the first byte is PROTOCOL_SOM_NOACK,  but the CI or the current index is always 0x90 and does not increment? s->curr_msg.len always gives 0x0a which I am expecting according to the following snipped of code:

` ups.msgToHover.SOM = 4 ;  // PROTOCOL_SOM; //Start of Message;
  ups.msgToHover.len = 7;   // payload + SC only
  ups.msgToHover.cmd  = 'W'; // PROTOCOL_CMD_WRITEVAL;  // Write value
  ups.msgToHover.code = 0x07; // speed data from params array
  ups.msgToHover.base_pwm = base_pwm;
  ups.msgToHover.steer = steer;
  ups.msgToHover.CS = 0;`

Here is the code I did change in the firmware, only config.h file : 
  #define CONTROL_TYPE USART2_CONTROLLED  // 
#define USART2PROTOCOLBAUD 115200
 #define USART2_BAUD 115200 

I did test the ASCII protocol and did work like a charm!  I am using ESP32 Wroom32  board.  Thank you for yoru help!

ahmed380, 2019-08-04:
more debugging here, I can confirm the serial buffer received by the hoverboard is corrupted. in protocol_byte function gets the SOM=0x4, which I expect, but the rest of data does not match what is being sent by the Arduino. For example ->curr_msg.len=232 but the correct length is 0xa. 

still not sure where the root cause of this issue is???

p-h-a-i-l, 2019-08-05:
Hmm, that's weird, I tested the example.
Did you try the other examples?

ahmed380, 2019-08-05:
I haven't tried the other examples as the pwm seemed to be the simplest one in terms of testing the comms between arduino and stm32.  I think, my issue might be related to the already open issue here
[https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/42](url)

ahmed380, 2019-08-05:
I can confirm that this issue was resolved by updating the following lines into config.h. Still not sure why?????

#ifndef SERIAL_USART_IT_BUFFERTYPE
  #define SERIAL_USART_IT_BUFFERTYPE unsigned char short
#endif
#ifndef USART2_WORDLENGTH
  #define USART2_WORDLENGTH UART_WORDLENGTH_8B 
#endif
#ifndef USART3_WORDLENGTH
  #define USART3_WORDLENGTH UART_WORDLENGTH_8B 
#endif

p-h-a-i-l, 2019-08-05:
But I can :/ 
Sorry for not pointing you in that direction. The default values were matched to hoverboard mode. The uart of the original sensor boards is a bit weird..
Additionally I did work only with presets in platformio.ini, so I haven't used config.h settings in a long time. 
Just today I merged the code fixing just that, so now the defaults should be good.

closed 2019-08-06 by ahmed380