bipropellant/bipropellant-hoverboard-api #9: Too much memory drained by Arduino library

ElectroAaron, 2019-10-28:
Hi, 
I'm trying to use an Arduino NANO (ATMEGA328P MCU at 16MHz) in order to write PWM, speed and position commands onto the board using machine protocol. The problem is that when I try to compile Arduino code, dynamic memory goes up to 321%, making it impossible to fit into the MCU. Is there any buffer or something on the library that I could remove in order to save enough memory??

Thank you!

p-h-a-i-l, 2019-10-31:
Hey,
If you just want to write some values, you could use very simple commands without using the api.
The API is just a wrapper for the protocol part of bipropellant, not size optimized.

closed 2019-11-05 by p-h-a-i-l

ElectroAaron, 2019-11-06:
Thank you for your answer!! I only need to write PWM, Speed and Position data from an Arduino NANO. Getting voltage data in the Arduino could be awesome, but it's not too important. Could you please post me an example of a data array in Bytes to send to the board?? I have been reading Wiki but I don't get all the stuff at all. Thank you so much.

p-h-a-i-l, 2019-11-06:
Have a look at https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/master/README.md#serial-machine-control 
you can find a simple example there.

ElectroAaron, 2019-11-07:
I've finally got it working. Thank you so much for your fast answers and for this amazing project.