bipropellant/bipropellant-hoverboard-api #13: Turning Hoverboard on or off

jnsw, 2020-04-10:
I can't find that, is it possible? Or is there a way to implement that? After a while my hoverboard starts beeping if no one is driving, so i want to turn it off.

p-h-a-i-l, 2020-04-10:
The command is not implemented at default, but it would be possible to to.

jnsw, 2020-04-11:
It would be enough if the hoverboard hardware can be switched off but the STM32 is still turned on, so the (maybe power consuming) high-pitched buzzing sound would disappear and everyone can remote start the hoverboard over the API. I power an ESP32 over the left sensor board cable with a voltage regulator, and it works fine for me, but I always have to turn it off or on manually from far away. 

I also can offer help, as I want to learn more about this software to gather more experience about microcontrollers and if you point out some ideas, where to touch the code, I maybe try my best.